davidphilipbarr/Sweep #1: Schematic and PCB are very different

xphoenixd, 2021-02-03:
Hi, I'm making an MX version of the sweep but the schematic and pcb are very different. The schematic has a matrix and diodes, but the switches on the PCB are direct pin, and with no diodes. Should I edit the schematic to remove the matrix and diodes?

xphoenixd, 2021-02-03:
I'll fix it anyway, it's too different to keep the same.

closed 2021-02-03 by xphoenixd

davidphilipbarr, 2021-02-04:
Yes! The sweep started as a 36 key corne hack and over time I've never
bothered to update the schematic until the on board ble I'm working on,
because I kind of had to :)

It's been something that has kind of grated on me for sometime, and I'm a
little ashamed of, but I am quite lazy.

I also made a mistake in the original which is why the upper pinky (q
normally) does not route very nicely. I've considered changing the pinout
but then I'd have to adjust the firmwares and I don't want to.

I'd love to merge the new schematic if possible!

Thanks

David