davidphilipbarr/Sweep #3: feature request: choc V2 supported board

deggers, 2021-06-19:
Hey,
Thank you soo much for your work creating a proMicro compatible ferris :) 

I wonder if you have any plans in creating a sweep V2 with mx spacing and choc V2 support? The corne light looks sooo cool with mt3 on the choc V2 - but I love the ferris layout .. what do you think?

davidphilipbarr, 2021-09-10:
i nearly have an mx spaced  board ready, i'm just finalising the footprint. mx/chocv1/chocv2 any others? alps?

amcguire2, 2021-09-12:
> i nearly have an mx spaced board ready, i'm just finalising the footprint. mx/chocv1/chocv2 any others? alps?

Thank you for working on this, I can't wait !

bazitov, 2021-09-25:
> i nearly have an mx spaced board ready, i'm just finalising the footprint. mx/chocv1/chocv2 any others? alps?

How do you look at the idea to have the hotswap and none-hotswap choc PCB like Architeuthis dux?

duckyb, 2021-12-21:
This issue was solved with the introduction of the **Sweep High**, it should be closed.

closed 2022-01-01 by duckyb