davidphilipbarr/Sweep #4: [Sweep v2] MCU pads very close to keyboard edge

duckyb, 2021-08-11:
Hi, I printed the latest version of the Sweep v2 from JLCPCB and I noticed a small issue with the design.
First of all, the fact that the PCBs are conjoined leaves two small bumps on the inner edge of the board, which can be sanded but might be risky with it being so close to the MCU's pads.
And second, the pads are very very close to the edge, nothing to worry about but it would be nice to have a slight margin between the MCU's pads and the edge of the PCB.

Both these issues could be fixed by adding a .5mm - 1mm margin to the inner edge of the PCB.

[image]

[image]

davidphilipbarr, 2021-09-10:
I think this fixed in the dev branch, the holes are very close to the edge still, but less so. 
[image]

duckyb, 2021-09-10:
It should be enough!

closed 2021-09-10 by duckyb