davidphilipbarr/Sweep #5: Move upper pinky key to outer column

michix, 2021-09-02:
I'm currently using a Kyria and basically reduced it to the size of the Ferris. There is just one thing, that I did different and which makes the layout extremely more usable: I moved the upper pinky keys to the position right and left, respectively, of the pinky home position. On QWERTY layout, I moved the `P` to the position right of the `;` key (i.e. the `'` key) on the right side. On the left side, I moved the `Q` key left to the `A` key (i.e. the `Capslock`).

From a usability point of view this is ingenious, although it makes the keyboard wider. What do you think of this modification? Probably it is just me not being able to reach the upper pinky columns comfortably...

davidphilipbarr, 2021-09-10:
i won't e doing this to the sweep, but you might be more interested in the hypergolic type keyboards. 
[image]

is this more like what you mean?

michix, 2021-09-10:
Thanks a lot for the reply and the suggestion. This is not really what I had in mind, butI guess my English is not sufficiently good enough to explain it properly, and probably my idea is also not that good and a more aggressive stagger on the pinky would solve my problem.I guess I will try the a.dux and see if this is good for me. Thanks again!

closed 2021-09-10 by michix