davidphilipbarr/Sweep #6: PCB Questions

kmplngj, 2021-09-04:
Hello,

I ordered some free Sweep-half-swept PCBs. Now Allpcb has some question to the files. Could you help me with this?

[image]

[image]
[image]
[image]

Thanks a lot

davidphilipbarr, 2021-09-04:
I've asked someone who got this printed at allpcb what they said. Off the
top of my head i think the first one can be left as pth and second is the
'crab' so I'd suggest leaving as is. If you sit tight I can get some
answers from someone with a working PCB from allpcb!

kmplngj, 2021-09-10:
I even got one more question. Perhaps someone owns if this needs to be fixed.

_Is the broken line pointed by the arrow normal?_
[image]

davidphilipbarr, 2021-09-10:
Hi! sorry for delay. that last trace shouldn't be there, but it's not doing any harm either! (i'll fix it)
No one else seems to have had any issues with allpcb, so i'd just suggest proceeding 'as is'