davidphilipbarr/Sweep #8: KLE json keymap file for Sweep34

humky, 2021-09-24:
I want to 3d print Sweep34 plates for handwired build. I usually use KLE (keyboard-layout-editor.com) and keyboardcad.com to produce the stl files for printing, but I can't find Sweep or Ferris KLE keymaps anywhere. (which is logical since it's PCB based boards) still, does anyone here have a KLE map for either Ferris or Sweep (I will surely mess up the angle of the keys or their placement if I try to reproduce it manually at KLE)

davidphilipbarr, 2021-09-26:
I know there is a ferris one, but I don't have a link to it. possibly your
best bet is to delete the keys you don't need from the kyria one?

humky, 2021-09-27:
I've been printing  and testing different plates for the past few days, and turns out I don't really need the arch on the bottom row that much. I'm equally comfortable with it being the straight row, maybe cuz I've been using Planck style keyboard for the past year or so. Anyway, here is my current board, I'm finalizing pinkey column stagger, all others pretty much perfect for my hands.

> [{x:2,a:7},""],
> [{y:-0.85,x:1},""],
> [{y:-0.9,x:3},""],
> [{y:-0.87,x:4},""],
> [{y:-0.93},""],
> [{y:-0.45,x:2},""],
> [{y:-0.85,x:1},""],
> [{y:-0.9,x:3},""],
> [{y:-0.87,x:4},""],
> [{y:-0.93},""],
> [{y:-0.45,x:2},""],
> [{y:-0.85,x:1},""],
> [{y:-0.9,x:3},""],
> [{y:-0.87,x:4},""],
> [{y:-0.93},""],
> [{y:-0.07,x:2},"","","",""]

closed 2021-09-27 by humky