davidphilipbarr/Sweep #9: https://jlcpcb.com won't print Sweep V2

danpassmore, 2021-10-03:
Hi, love this design, but when I attempted to print the Sweep V2 using https://jlcpcb.com, they wouldn't print the board for the following stated reason:

Quote:
Hi there,so sorry that the width of the breakaway tabs is too small which may cause board breaking off very easily.Please kindly increase it to 4mm at least.You can click "replace file" button to re-upload the updated gerber file in your account.Thanks.

Have you ever successfully used jlcpcb to print the Sweep V2?
Is what they're suggesting a reasonable request?

danpassmore, 2021-10-03:
Okay, I was able to work around their audit rejection by specifying the PCB material FR4 instead of Aluminum.  Seems something about printing in aluminum puts more stringent constraints on the breakaway tab width.  

(I've never ordered a custom pcb before, so not sure if aluminum is a good idea or not, but my thought process was that it might result in a nicer/sturdier keyboard, and it was even a dollar cheaper than FR4!  Oh well.)

closed 2021-10-03 by danpassmore

vogler, 2022-02-10:
> (I've never ordered a custom pcb before, so not sure if aluminum is a good idea or not, but my thought process was that it might result in a nicer/sturdier keyboard, and it was even a dollar cheaper than FR4! Oh well.)

Just wondered the same...