davidphilipbarr/Sweep #14: What are the differences between Sweep v2 , Sweep-half-swept and Sweep-Compact?

IronArthur, 2021-11-11:
I´m trying to decide what Sweep to build. My ultimate goal would be that at least the left board is wireless when nice!nanos are available. At first i will use some pro micros.

I'm having trouble finding the differences between those 3 pcbs, especially i can´t find what are the differences of the half-sweet

Sweep v2: has wireless capactity with holes for the battery? and on-off button i think
Sweep-half-swept: the same as v2 but the pcb is cheaper because it is smaller (8€ cheaper on jlpcb). Cons: right board has pro micro upside down? you end with 2 and half kits of pcbs (instead of 5 PCBs in V2)
Sweep-compact: cheaper pcb as the previous one, but no battery holes and no tenting??

If i understand everything, there is no point of using the compact when the half-swept exists, but maybe there are some other differences that I am not aware of.

duckyb, 2021-11-11:
A few facts that may clear some things up:
- All versions support wireless. If there are no battery pads you simply connect the red wire to RAW and the black wire to GND on the micro controller. The battery pads are just extensions of those pins.
- If you want to build a wireless board both sides must use a nice!nano
- Contrary to the sweep compact, the half swept has a double micro-controller footprint, letting you install both daughter-booards facing the same way. (both with components facing up / down)

| Device | Bluetooth Support<sup>[1]</sup> | On/Off Switch | Reversible PCB | Choc V1 | Choc V2 | Choc Mini | MX & Alps | Choc Spacing<sup>[2]</sup> | Tenting<sup>[3]</sup> |
| :--- | :---: | :---: | :---: | :---: | :---: | :---: | :---: | :---: | :---: |
| Sweep2            | ✔ | ✔ |   | ✔ |   |   |   | ✔ | ✔ |
| Sweep Half Swept     | ✔ | ✔ | ✔ | ✔ |   |   |   | ✔ | ✔ |
| Sweep Compact     | ✔ |   | ✔ | ✔ |   |   |   | ✔ |   |

IronArthur, 2021-11-11:
Thanks for the fast answer!

I thought that for wireless connection you could connect wirelessly to computer (instead of USB cable) and connect with trrs to the other board. So if I understand correctly, both boards connect independently to the computer? Good to know.

On kicad i´ve seen that the Half Swept has a couple of holes on the topside of the pcb (reset + gnd) that v2 doesn´t have, what are those ( on/off switch relocated?)
[image]

I still don´t get what´s the point of the compact when the half-Swept exists.

duckyb, 2021-11-11:
> So if I understand correctly, both boards connect independently to the computer?

Yes.

> what are those

I have no idea.

> I still don´t get what´s the point of the compact when the half-Swept exists

@davidphilipbarr doesn't like to delete old stuff! The only dis-advantage is that the half-swept is marginally wider because of the double footprint

kuroahna, 2021-11-11:
@duckyb Thank you for the very helpful information! It'd be great if we can update the README with it.

Also, I'd like to double check, but if I want to build a wired Sweep Half Swept instead of Sweep v2 from AllPCB, will your video still apply? https://www.youtube.com/watch?v=fBPu7AyDtkM

I'm looking to use:
1. [2x Elite-C Low Profile v4 microcontroller](https://splitkb.com/collections/keyboard-parts/products/elite-c-low-profile-rev4-microcontroller) installing both face down on the half swept
2. [4x Mill Max Low Profile Sockets with 48 Mill Max Pins)(https://splitkb.com/products/mill-max-low-profile-sockets?variant=31945995845709)
3. 2x Reset switch
4. 2x TRRS Jack
5. 34 PurpZ switches
6. 34 keycaps
7. Rubber feet

duckyb, 2021-11-11:
@kuroahna I already updated the readme in #15.

Yes you can still follow the video with your half swept board. Only thing that I have to warn you about is that the Purpz might be a very tight fit in the PCB because of different tolerances. (I never used them but I've read some discord messages where a user had some problems installing purpz on a sweep)

With that being said, please only use the issues page to discuss problems with this project or request new features! If you have questions about my tutorial or need help with your build, the youtube comments or the discord server are a better place to ask.

IronArthur, 2021-11-11:
What's the point of the "Reset switch"?, it´s probably the thing that goes on the holes in the image i've attached, but what's suppose to be? just a button??

duckyb, 2021-11-11:
> What's the point of the "Reset switch"?, it´s probably the thing that goes on the holes in the image i've attached, but what's suppose to be? just a button??

Yes. It's just a small button that shorts GND and RST pins. You can just use tweezers or any other conductive object to short them instead.

IronArthur, 2021-11-15:
Thanks for all the info! i´m waiting for my parts! and the pcb from allpcb free prototype.

closed 2021-11-15 by IronArthur