davidphilipbarr/Sweep #16: Sweep High Gerbers

telepuzikIsmail, 2021-11-13:
There seems to be something wrong with the gerberfiles for the Sweep High. 
They dont work on JLCPCB, no preview and so on.. or am I doing something wrong?

duckyb, 2021-11-13:
[image]

It might be caused by the new footprint that has some overlaps. The lower two overlaps on the drill layer are reported as errors by the DRC tool in Kicad.
If you still order it the JLC team will still check it for you and let you know if there are issues with your order.

Keep in mind this is a new design and I don't even know if it was ever ordered by anyone before

telepuzikIsmail, 2021-11-13:
Thanks for the quick answer.
I have order boards now, I'll let you know if it worked out or not.

Micosilent, 2021-11-15:
I've tried to order Sweep High boards as well, and JLCPCB is complaining about the minimum breakaway tab being 4m
[image]

duckyb, 2021-11-15:
[image]
[sweep-high_229x356mm_for_JLCPCB.zip](https://github.com/davidphilipbarr/Sweep/files/7537976/sweep-high_229x356mm_for_JLCPCB.zip)

On 13/11/2021 @davidphilipbarr shared a gerber for JLC that is missing in this repository. I'm reposting it here for people that might be interested in it.

telepuzikIsmail, 2021-11-15:
This is what they wrote me in the first mail:

> Sorry that the minimum width of breakaway tab with mouse-bites we recommend is 5mm.
> Sorry that the minimum width of breakaway tab we recommend is 4mm.
[image]
[image]

After replacing the gerbers with those from the post above they wrote:

> the spacing from slot to slot and slot to board edge are too small which will cause the board broken easily,please kindly increase it to 4mm at least.
[image]

Micosilent, 2021-11-15:
The thing is that we literally want the board to break, so the two halves come apart. If you want to wait until I receive mine, I have placed the order and told them to disregard the <4mm breakaway, and will update when the order arrives.

Otherwise just tell them to go ahead with the fab

duckyb, 2021-11-15:
@Micosilent @telepuzikIsmail the mouse bite footprint is the same used on the Sweep2 PCB and there are no issues with that one. I ordered them before and they didn't ask me anything and the PCBs were ok.

I guess they just want to complain more this week.

Micosilent, 2021-11-15:
JLCPCB is refusing to make the pcbs with the small mouse bites 😞

telepuzikIsmail, 2021-11-16:
I wrote them to go ahead and make the pcbs the way they are now and they decided not to answer me so I canceled the order...
I'll try to edit the pcb this weekend and try again.

duckyb, 2021-11-16:
@telepuzikIsmail @Micosilent I tried to do it for you.

| Current version | My updated fork |
| --- | --- |
| [image] | [image] |
| [image] | [image] |

My fork gerber: [sweep-high-gerber.zip](https://github.com/davidphilipbarr/Sweep/files/7546704/sweep-high-gerber.zip) 

Please note that I removed the GND-GND wire that crossed the mouse bite, as it didn't fit through the drill holes. But I don't think it is important as it will be broken off.

Also note that this gerber does not show a preview on JLCPCB, just like the gerber from the current version in this repo.

microfortnight, 2021-11-18:
JLCPCB accepted my order based on the fork of @duckyb 
Didn't use the linked gerbers because I wanted to remove the crab 😜
I used asukiaaa/gerber_to_order to create the gerbers and the preview on JLCPCB is displayed correctly.

duckyb, 2021-11-18:
@microfortnight thank you for the feedback! If you could update us once more after receiving the PCBs it would be awesome.
In particular regarding the breakaway tab; if it snaps well, how much clean up is required, possible improvements, etc.

Also nice tip on using that gerber tool. I'll check it out!

nodupe, 2021-11-18:
I’ve placed the order a few days ago and had the same issue, even after assuring them it was supposed to break, they still complained. Just uploaded the new file now, looking forward to their answer.

closed 2021-11-19 by davidphilipbarr

Narice, 2021-11-30:
@microfortnight, did you get the board? I'll probably build one and having some feedback to know if this gerber file worked would be really awesome :grin:

microfortnight, 2021-12-08:
@duckyb PCBs look fine, didn't build the board yet.
Seperating the halves is easy, one half can be broken up by hand, you will need pliers to separate the tabs from the other half. That's a very minor disadvantage, compared e.g. to the crkbd cherry, where it's possibe to do it using hands only.
Apply a very light sanding and it is good to go :+1:

lxhan, 2021-12-11:
Hey guys, can someone please explain or show how to plug in switches into sweep high board? I've just received pcb but wanted to double check before soldering, is it ok that mx switches are loose compared to what I saw in @duckyb video on youtube or something wrong with my board?

duckyb, 2021-12-11:
@lxhan Yes it's somewhat normal. But make sure that you are using 5 pin switches (PCB mount)

Narice, 2021-12-11:
@microfortnight thanks! that really helps

commit 216f4de3 (2021-11-19): should fix #16