davidphilipbarr/Sweep #17: sweep 2.1 gerbers

mybigman, 2021-11-16:
gerber zip is missing for 2.1 or do we use the one from 2

thanks

Micosilent, 2021-11-16:
2.1 is Work in Progress, you can use the gerbers from v2  though :D

duckyb, 2021-11-23:
The gerber for 2.1 is available if you want to try it!

eXsoR65, 2021-11-29:
@duckyb looks like the gerber file is not reflecting the change of removing the extra pin on one of the switched on the right half. I tried to generate the gerber files but gave me a warning about filling in. But I'm new to this so not sure if I should or not. Any help on this is appreciated.

duckyb, 2021-11-29:
@eXsoR65 you can find the updated files on my branch https://github.com/duckyb/Sweep/tree/update-gerbers
I also created #24 to update them here.

eXsoR65, 2021-12-02:
@mybigman Issue was address so you can close the issue. see PR #24

closed 2022-01-01 by duckyb