davidphilipbarr/Sweep #20: Reset switch part number

ch604, 2021-11-19:
From the images they appear to be SMD 3x4x2 2 pin pushbuttons, correct?

duckyb, 2021-11-23:
[image]

Documentation: https://omronfs.omron.com/en_US/ecb/products/pdf/en-b3u.pdf

This component is compatible. Part number is **B3U-1000P(M)**
(this is just one of many compatible buttons)

duckyb, 2021-11-23:
[image]

This is the footprint, for reference

ch604, 2021-11-23:
Thanks for that pcb breakout and the exact reference part, looks like the piece I selected will fit!

closed 2021-11-23 by ch604