davidphilipbarr/Sweep #22: Sweep 2.1 has 2 designs?

paulchiu, 2021-11-25:
First of all, thanks for the great work on the Sweep, I discovered this project through Ben Vallack's YT video.

Not sure if this is the right place to ask, so feel free to close.

I have placed an order with JLCPCB with the 2.1 Gerber file and received this message from them:

> You need to pay additional for your order.
> 
> 1.The order info shows that there is only one design , but actually, there are 2 different designs in your file, so there will be an extra cost for it if you want to make it as show in your file.Please kindly choose the correct designs when you place your order next time. 

I set my order details based on settings recommended by the Kyek video; although I did deviate by using the 2.1 file.

I noticed that https://github.com/davidphilipbarr/Sweep/issues/17#issuecomment-970198629 mentioned that 2.1 is still WIP, so my question is will my order produce a functional PCB or do I need to make another order with the v2 Gerber file?

I wasn't sure what I was looking at with the zip file in Kikad, but upon closer inspection, the only thing I can see that seems like a secondary design is the `sweepv2.1_plate` file, is that what JLCPCB would be referring to?

davidphilipbarr, 2021-11-25:
> I noticed that #17 (comment)

It's probably just because it has detected the two slides, so it's saying
there are two boards in one design, which is true.
JLC can be a bit weird and inconsistent about this kind of thing,
usually the extra charge is small though.
The 2.0 had a trace joining the halves, so it might avoid this warning. If
in doubt i'd use the 2.0 as it's been more tested, but it should be fine
and the warning from jlc is to be expected really.

hope that helps!

duckyb, 2021-11-25:
The v2 also doesn't have this trace, I removed them with the new breakaway tabs. If this trace is needed the breakaway tab footprint needs to be updated.

paulchiu, 2021-11-25:
Thanks for the quick clarification guys.

> usually the extra charge is small though.

Yes, it was only like $5 AUD so I just paid it as soon as I got the email, just wanted to confirm here that the design was ready for use; thanks for confirming that it is.

> If you don't want to wait for an update or pay the extra cost you can order the half-swept version instead, which is just one design.

It's not a big deal really.

> the warning from jlc is to be expected really.

Do you mind if I do a PR for the 2.1 README with the information you guys provided in this issue?

PR #23 (2021-11-30): doc(sweep2x): Added PCB ordering notes to README
Adds some PCB ordering settings to prevent confusion.

Resolves #22.
closed 2021-11-30 by davidphilipbarr