davidphilipbarr/Sweep #30: Question: Difference between High and High M

isavcic, 2021-12-26:
Hello,

I apologize if this question is an obvious one, but I'm about to embark on building an MX-version board for myself. Which version of High should I use? What is the difference?

Thanks!

duckyb, 2021-12-26:
High M is the reversible version of the high. Like sweep2 vs sweep half swept.
You can use both but the high is more tested

isavcic, 2021-12-26:
In which way "reversible", as in the board can be on the bottom side of the board?

duckyb, 2021-12-26:
@isavcic yes. Reversible means that the same PCB is used for left and right side. In the regular sweep high you have a dedicated PCB for each side.
This is just a minor difference, both devices function the same.

isavcic, 2021-12-26:
@duckyb Thanks!

closed 2021-12-26 by isavcic