davidphilipbarr/Sweep #33: miryoku keymap

rnyPlanet, 2022-01-04:
is there a keymap example? I want to make a keymap according to the [example ](https://github.com/rnyPlanet/qmk-crkbd-keymap-miryoku/blob/main/keymap.c#L43),  but I can’t find anything and figure it out [_**in**_](https://github.com/qmk/qmk_firmware/blob/master/keyboards/ferris/sweep/sweep.h#L31)

**really hard to figure it out**

duckyb, 2022-01-04:
Miryoku on a 34 key device would be like the 36 key version but the inner thumb key is emulated by a combo of both thumb keys.
Source: https://github.com/manna-harbour/miryoku/issues/56

If you don't want to wait for the official implementation you can do it yourself by starting with the default miryoku and adding the combo support for the third thumb key.

rnyPlanet, 2022-01-05:
> Miryoku on a 34 key device would be like the 36 key version but the inner thumb key is emulated by a combo of both thumb keys. Source: [manna-harbour/miryoku#56](https://github.com/manna-harbour/miryoku/issues/56)
> 
> If you don't want to wait for the official implementation you can do it yourself by starting with the default miryoku and adding the combo support for the third thumb key.

оh. well, the fact that there are fewer keys is not scary, I cannot understand - if I can add [keymaps lines](https://github.com/rnyPlanet/qmk-crkbd-keymap-miryoku/blob/main/keymap.c#L34) to [this](https://github.com/qmk/qmk_firmware/blob/master/keyboards/ferris/sweep/sweep.h#L47) file? and correct a bit

 [These](https://github.com/qmk/qmk_firmware/blob/master/keyboards/ferris/sweep/sweep.h#L47) L01 and others knocked me down a bit and I can't find an explanation on the Internet

duckyb, 2022-01-05:
You should change that file only if you changed the physical layout of the keyboard (hardware modification)

If you want to create a new layout just make a new keymap folder

closed 2022-01-05 by rnyPlanet