davidphilipbarr/Sweep #37: Top Plate Generation for Sweep High

ThePreviousOne, 2022-01-29:
How was the top plate for this created? I added a key to the design in KiCad but the cutouts I used for my design flie were too big. I pasted it over top of the Sweep's plate file and found that the holes don't even line up. This makes me wonder if the Sweep High plate even really fits, I'm sure it does but I don't know how else to explain this

davidphilipbarr, 2022-01-29:
The plate files are generated from the the main pcb with this footprint swapped for the switches

[SW_PG1350_cut.zip](https://github.com/davidphilipbarr/Sweep/files/7964082/SW_PG1350_cut.zip)

(and then the edge cuts modding)

The dxf and svg are exported from this. have look, let me know if you think what is there does not match up with the output from the plate files i have just uploaded.

good luck :)

ThePreviousOne, 2022-01-30:
Thank you so much, that worked, now I can finally finish my kb

closed 2022-01-30 by ThePreviousOne