davidphilipbarr/Sweep #40: Incorrect Board outline

John-Ingram, 2022-03-02:
Some there are some issues in the board outline (overlapping outline) for at least the Sweep-high board (although I expect that the issue affects the other boards also). 

This issue can be demonstrated by trying to export the board in the IDFv3 format. If you open the generated.emn file in a text editor, you'll notice that there are only a few lines that start with zero. These lines are the coordinates for the board outline, and there should be more of them. 

The fix for this issue is to go through the board outlines and check every segment of the board outline to make sure there are no errors.
I will do this for the Sweep-high board sometime soon, but I figured I'd let everyone else know until then.

JiaGengChang, 2022-03-16:
I have tried to build a sweep high, and a lot of the keys on the right side are not working. Is that related to this issue?

duckyb, 2022-03-17:
@JiaGengChang No

duckyb, 2022-04-24:
Fixed in https://github.com/davidphilipbarr/Sweep/commit/972bf728dd9374cb81e3659c3c51c807ad7520d0 but some more fixes are needed before I can merge / update the gerbers

duckyb, 2022-04-25:
Merged https://github.com/davidphilipbarr/Sweep/commit/f87ba64e6f716add1bc7cf43377e1d57bf9c240c

closed 2022-04-25 by duckyb