davidphilipbarr/Sweep #42: Strange behavior on the left hand board

plattfot, 2022-03-13:
This is probably not the right place to ask and feel free to close if it is not.

I assembly one Ferris Sweep v2.1 after ordering 5 PCBs from JLCPCB (thank you for the walk through on that).

Everything was straight forward, I only had one issue with one of the PCB's hole for the micro controller. Two of the holes was too small so the micro controller didn't fit. But the second board I tried worked out perfectly.

The main issue I have is after a while a key on my left hand will stop working. Mainly this will be `c`, but I have seen `w` and `t` as well. The strange part is that if I press any other key on the left hand the dead key will be pressed and stay pressed until I disconnect the keyboard.

First I thought it might be because of bad soldering. But I've re-solder all switches, pins for the trrs and micro controller.

I have checked with a multimeter for shorts and if the switches work. Everything looks good. For example if I remove the micro controller, hook up the multimeter to pin 23 (GND) and pin 7 (Switch 13 aka `c`). Then press `c`, the multimeter register the press. And no other key triggers it.

I have swapped out the Elite-C micro controller for a Pro micro I had in another Keyboard. Same issue.

I don't see any damage on the PCB.

Is this a bug in `qmk`

duckyb, 2022-03-13:
Check this, and skip the part about diodes:

https://docs.splitkb.com/hc/en-us/articles/360010552339-One-or-more-keys-don-t-work

plattfot, 2022-03-13:
Thanks for the link.

I did do a second pass today and notice some flux residue on the front of the board. I had some issues with soldering one of the legs for the choc switches so I had used some extra flux I had lying around to solder them properly. That looks to be my mistake though. As after I had desoldered one of the switches, the bottom of it was covered in flux residue. So I just desoldered all the switches, cleaned them, moved everything except the low profile sockets to a new board. In case there are other issues with that particular pcb.

And now everything seems to be working. :crossed_fingers:

My guess is the flux that pored down the legs of the switches, caused a short. Which also explain why it worked sometime and why different keys stopped working.

Lesson learned:
- Don't use extra flux when soldering through hole.
- Use a better tip when soldering. I changed to another tip and that made it much easier to solder those legs.

And I'm glad I socketed the micro controllers. Made it much easier to debug and I didn't have to desolder one of them when changing board. 

And thank you for making the pcb open source! Was pretty straight forward to just open kicad and see where the traces are on the board. 

Now I'm going to enjoy my ferris sweep!

closed 2022-03-13 by plattfot

duckyb, 2022-03-14:
Good to hear you were able to fix it, have fun!