davidphilipbarr/Sweep #43: Which is the behaviour of the on/off switch?

leiserfg, 2022-03-23:
I got a preassembled sweep with nice!nanos but it seems like the switches don't turn it off, is that the intended behaviour or is it incorrectly mounted?

davidphilipbarr, 2022-03-24:
sounds like something is wrong there, do you have some photos of the
switch/battery connection?

leiserfg, 2022-03-24:
[image]
[image]

leiserfg, 2022-03-24:
[image]

leiserfg, 2022-03-24:
I took the picture with the switch on and off but I can type with it no matter what

duckyb, 2022-03-24:
@leiserfg How can the switch do anything if the battery is directly connected to the nice!nano?
In order for it to work, the battery must be connected to the sweep PCB.

[image]

duckyb, 2022-03-24:
In case you don't have the tools to fix it, I recommend using these settings in your cradio.conf

```py
# enable deep sleep support
CONFIG_ZMK_SLEEP=y

# the keyboard disconnects from bluetooth.
# uses very little power, but it may take
# a few seconds to reconnect after waking.
# 1.800.000 ms = 30 minutes
CONFIG_ZMK_IDLE_SLEEP_TIMEOUT=1800000
```
(adjust timeout value according to your preference)

closed 2022-03-24 by duckyb

leiserfg, 2022-03-24:
Thanks a lot, I supposed it was not well wired but didn't know the correct way to do it, I'll recompile the firmware that way and try to rewire it once I get a good iron.