davidphilipbarr/Sweep #45: Sweep2 vs Sweep Bling MX

p0da, 2022-03-31:
Just looking for a quick clarification on which type I should build as while the Sweep2 is currently recommended, the Sweep Bling MX seems to be just an overall better version of the Sweep2. Are there any drawbacks to the Bling MX that the 2 doesn't have? Thanks for the help, kinda new to the space.

duckyb, 2022-03-31:
[image]

The main difference is that one is for choc switches, the other is for MX switches.

p0da, 2022-04-01:
That clears up my question perfectly, thanks for the help. And that chart is really useful might want to put it on the README.

closed 2022-04-01 by p0da