davidphilipbarr/Sweep #47: Reset button for the compact low version

riveraletelier, 2022-04-23:
First, thanks for putting all of this together.

There is a small issue with the description. It recommends the Omron B3U-1000P(M) for the reset button. It works well, expect for the compact low version that is designed for a through hole button. It would seem that the Apem MJTP1243 button would work, but I haven't tried it yet. Here is a link with to the datasheet in Digikey:
https://media.digikey.com/pdf/Data%20Sheets/APEM%20Components%20PDFs/MJTP%20Series-6MM.pdf

duckyb, 2022-04-23:
Thanks for reporting this issue. I will replace the footprint on the compact low to use the `Omron B3U-1000P(M)` component instead.

riveraletelier, 2022-04-28:
That would be great, thanks!

duckyb, 2022-05-19:
Fixed and merged

closed 2022-05-19 by duckyb