davidphilipbarr/Sweep #48: Sweep High Plated Slot Width

ballin2much, 2022-04-23:
Trying to purchase sweep high from JLBPC and they are saying the following:
"the minimum width of plated slot need to be 0.5mm or more, could we enlarge the width of 0.3mm and 0.4mm slots to 0.5mm?"
 [image]

Very unfamiliar with PCB design so I'm unsure if this is an issue with the gerber file, if I did something wrong, or if I just need to tell them that it is fine to enlarge.

duckyb, 2022-04-24:
@ballin2much Reply:

```
Yes, you can enlarge the slots to 0.5mm.
Thanks.
```

This is the second time they complain, we should change the footprint.

ballin2much, 2022-04-24:
Thanks for the quick reply. Appreciate all the documentation/tutorials.

closed 2022-04-24 by ballin2much

duckyb, 2022-04-24:
[image]

Updated the silk screen and enlarged the slots

duckyb, 2022-04-25:
This issue is now fixed: https://github.com/davidphilipbarr/Sweep/commit/f87ba64e6f716add1bc7cf43377e1d57bf9c240c
The gerber is updated to the new version.