davidphilipbarr/Sweep #55: My keyboard only works the half connected to the usb

rodrigocam, 2022-05-20:
Hey, I build a sweep keyboard v2 but it only works the half connected to the USB. I have a TRRS cable but it didn't work. I have tested the continuity of the holes of the TRRS jack in the PCB and I found a strange behavior, the two holes in the picture below give continuity, is it right?

[image]

davidphilipbarr, 2022-05-20:
They should both be gnd, so that is normal. Do you have some pics of the
top and bottom of your board?

rodrigocam, 2022-05-20:
Yes, the one assembled and the other factory new:

[image]
[image]

[image]
[image]

rodrigocam, 2022-05-20:
1. Test that all pins on the VCC net are connected ✅
2. Test that all pins on the switch 18 net are connected ✅

duckyb, 2022-05-20:
> 1. Test that all pins on the VCC net are connected ✅
> 2. Test that all pins on the switch 18 net are connected ✅

Then it shouldn't be a hardware issue 🤔

rodrigocam, 2022-05-21:
Is it possible I'm flashing the wrong firmware? I'm using this one https://config.qmk.fm/#/ferris/sweep/LAYOUT_split_3x5_2

duckyb, 2022-05-21:
@rodrigocam try to flash eeprom right/left on your respective micro controllers
`QMK Toolbox > Tools > EEPROM > Set L/R`

daskygit, 2022-05-21:
You're likely going to want to compile firmware with a option enabled.

Try add `#define SPLIT_USB_DETECT` to your config.h and flashing both sides.

This changes the USB detection so it doesn't use VBUS sensing which can be a problem with some promicros.

rsteube, 2022-05-21:
Second on the [SPLIT_USB_DETECT](https://github.com/rsteube/qmk-keymap/blob/master/ferris-rgaertner/config.h#L8). Coincidentally also had the issue today where it seems the promicro always tried to be master.
I think sth. is up with  [J1](https://docs.splitkb.com/hc/en-us/articles/360010588860-Only-one-half-of-my-keyboard-works-at-a-time-but-not-when-they-are-both-connected) which apparently is bridged although the pads are clean.
Adding the config fixed it.

rodrigocam, 2022-05-23:
> You're likely going to want to compile firmware with a option enabled.
> 
> Try add `#define SPLIT_USB_DETECT` to your config.h and flashing both sides.
> 
> This changes the USB detection so it doesn't use VBUS sensing which can be a problem with some promicros.

This worked like a charm, thanks! 👍

closed 2022-05-23 by duckyb

mladedav, 2023-04-11:
I've just run into the same issue. Is there a downside to using this option or can I try and add this option to the qmk firmware?

cgahr, 2024-02-13:
I have the same issue. However, for me, `#define SPLIT_USB_DETECT` does not solve the problem. 
VCC net and switch 18 net are fine for me too. 

Any help would be appreciated!

duckyb, 2024-02-13:
> I have the same issue. However, for me, `#define SPLIT_USB_DETECT` does not solve the problem. VCC net and switch 18 net are fine for me too.
> 
> Any help would be appreciated!

1. Make sure you are using a TRRS cable and not a TRS.
2. Connect the cable and test net 18 with the cable connected. Also test that parts that should be on different nets are not connected (shorts).

cgahr, 2024-02-14:
> 1. Make sure you are using a TRRS cable and not a TRS.
>
>     2. Connect the cable and test net 18 with the cable connected. Also test that parts that should be on different nets are not connected [...]

1. I'm using a TRRS cable
2. VCC and net 18 both look good, with cable connected
3. I couldn't find any short 
4. each half individually is working fine

I compile with the following command
```
qmk compile -kb ferris/sweep -km ryoku -e MAKECMDGOALS=uf2-split-left -e TARGET=sweep_ryoku_left
qmk compile -kb ferris/sweep -km ryoku -e MAKECMDGOALS=uf2-split-right -e TARGET=sweep_ryoku_right
```
and 
```
# rules.mk
CONVERT_TO=elite_pi
SPLIT_KEYBOARD = yes
CONSOLE_ENABLE = yes
```
```
\\ config.h
#define SPLIT_USB_DETECT
#define SERIAL_DEBUG
```

When I run `qmk console` and conntect my keyboard, I get:
```
DPB:Ferris sweep:1: SPLIT: receiving handshake failed
DPB:Ferris sweep:1: SPLIT: receiving handshake failed
DPB:Ferris sweep:1: SPLIT: receiving handshake failed
DPB:Ferris sweep:1: Failed to execute slave_matrix
DPB:Ferris sweep:1: Target disconnected, throttling connection attempts
DPB:Ferris sweep:1: SPLIT: receiving handshake fail
```

duckyb, 2024-02-15:
@cgahr I would recommend taking this to [QMK's discord chat](https://discord.gg/Uq7gcHh) as I don't know how to help you and not many people will be looking at this page.

cgahr, 2024-02-15:
Thank you anyways :)