davidphilipbarr/Sweep #56: Sweep Compact Low  (additional keys)

zaddok, 2022-05-24:
It looks like the “Sweep Compact Low - The only version that supports Choc minis.” Would be my ideal keyboard, but with an couple of extra custom keys. (Maybe three keys added on each end (6 keys), or two keys added for each thumb (4 keys))

I was thinking of trying to learn how to to do that, but I wanted to check if anyone had done anything like this before.

duckyb, 2022-05-24:
If you want to **not** use diodes / switch matrix you can only have 1 key per switch pin. This puts a limit of 17 keys per side with a pro-micro. You can get extra keys only if you use a switch matrix with diodes, or if you use an elite-c and make use of the extra pins.

davidphilipbarr, 2022-05-24:
The sweeeeep uses the extra pins from the elite-c for leds and a screen:

https://github.com/sadekbaroudi/sweeeeep

the cantor uses a black pill to get the extra pins:

https://github.com/diepala/cantor

They might be of interest?

closed 2022-05-25 by duckyb