davidphilipbarr/Sweep #61: Ferris sweep on nRFMicro

conceptroot, 2022-09-10:
Hi guys. 
Is it possible to use nRFMicro instead of Nice!Nano in Ferris Sweep?
It is almost impossible to get Nice!Nano in my country.

If ferris sweep runs on nRFMicro, it would be great to indicate it in documentation.

GerardoNevarez, 2022-09-10:
there's no reason why it should not work... just need to make sure the headers are slightly more open, in case the eByte module bumps into them.

See the module on top, and how close the headers will be with the nrfmicro upside down.  just open them a bit to the sides.  [nrfmicro with shield removed ](https://user-images.githubusercontent.com/852547/136052828-7e8c5e5a-bf60-4805-b5f9-ffaaf6725221.jpg)

duckyb, 2022-09-10:
| header | pcb clearance |
| --- | --- |
| [image] | [image] |

closed 2022-09-11 by duckyb

conceptroot, 2022-09-11:
Thanks for replies guys.
It should work I guess. I asked a guy ufan (https://github.com/ufan/sweepro-ce), and he has regular sweep on nRFMicro.
Here is a picture of regular sweep on nRFMicro:
[image]

I ordered all PCBs in JLCPCB and in couple of month I'm gonna try this build (nRFMicro on Sweep LP). 
I'll let you know if it works.

duckyb, 2022-09-11:
@conceptroot good luck! 🍀