davidphilipbarr/Sweep #63: Feature request: Battery JST connector

shrekale, 2022-11-24:
We need JST connector for battery connection instead of soldering wires on there.

davidphilipbarr, 2022-11-25:
Need is a very strong word :)

There is a version where the jst connector sits over the trrs, as you would
never use both together.

I'll see if i can dig it out

shrekale, 2022-11-25:
I am actually working on replacing TRRS with that JST, With NiceNanos we don't use it anyway, those MCs can't communicate through TRRS. If there's a branch that has that, please let me know, or I will do some KiCAD work & get it done for myself.

shrekale, 2022-11-25:
If you can't find, I will push my version named properly in one folder, others will use it then.

closed 2022-11-25 by shrekale

davidphilipbarr, 2022-11-25:
https://github.com/davidphilipbarr/testing-boards/tree/main/Sweep%20v2.2-jst

This is the version I have, but unless someone else tests it it will remain
in limbo.

There is also an older sweep with the trrs removed.

https://github.com/davidphilipbarr/testing-boards/tree/main/Sweepv2.1-nt

Feel free to add yours too :)

cheers

David

shrekale, 2022-11-25:
Nice. Thanks! Let me see this & if mine's better, I will check it in. I can see why it's not just one version of the PCB & there are subfolders, people might not want it wirelessly, they might need TRRS.