davidphilipbarr/Sweep #65: Dimensions for Bling MX switch plate/back plate unclear

astralbat, 2022-12-07:
Hello,
When I was ordering the switch plates and back plates, I was not sure if these were the same dimensions mentioned for the PCB. I assumed they were, but the uploaded gerber for the switchplate has a different dimension in the filename.

In the end, when ordering, I assumed they were all the same dimension as the PCB and that the filename has a typo.

This is particularly a problem with the swithplate as JLCPCB did not autodetect them or render the result.

Thanks.

duckyb, 2022-12-07:
The files have been ordered successfully before, so I don't think you need to worry about it.
I will check if the dimensions in the filename are correct or not, all I can say for now is that they are auto-generated by a plugin.

duckyb, 2022-12-07:
Also, the dimensions on the quote page are only needed to calculate the cost of the order. Their machines will reference what's in the file to make the PCB.

astralbat, 2022-12-07:
I see, it was my assumption the dimensions would scale the board. I think it would be good to clarify a little more on the readme. Thanks.

closed 2023-01-31 by duckyb