davidphilipbarr/Sweep #66: Add switch plate and bottom plate for the Sweep High M

isacben, 2022-12-11:
Hi,

It would be nice to have switch and bottom plates for the Sweep High M version.

The switch plate would be specially important because the Sweep High M version requires Kailh switches with no alignment pins, which makes it difficult to solder them aligned.

Thanks!

duckyb, 2022-12-11:
I think the bling mx plate might be compatible with it

isacben, 2022-12-11:
Thanks, Duccio,

I was wondering about that, and I printed both PCBs on paper. It looks aligned, but I cannot confirm that it works. Would have to order the PCBs.

duckyb, 2022-12-12:
The switch plate will work. The only problem is that it will cover the reset button location.
The bottom plate PCB will not work with it because the main PCB is missing the holes for mounting hardware.

If someone wants to add those holes in the correct location feel free to submit a pull request.

closed 2022-12-19 by duckyb