davidphilipbarr/Sweep #67: Standoff/Spacer Holes for Sweep Bling MX/LP are too tight

hmngwy, 2023-01-31:
I'm using M2 hex spacers and they have an outer diameter of 3.0mm. M2 knurled spacers have an outer diameter of 3.1mm. The hole looks like it measures 3.2mm, however due to fab tolerances they come out around 3.1-3.2mm, they are a tad bit too tight, with some pressure they will go in, but I think 0.1mm (min) to 0.2mm (max) increase in diameter would fix this issue.

duckyb, 2023-01-31:
The diameter was chosen for this type of spacer, which measures 3.00mm in diameter. Are you not using this type of spacer?

[image]

duckyb, 2023-01-31:
These are the tolerances according to [JLC's capabilities documentation](https://jlcpcb.com/capabilities/pcb-capabilities)

[image]

Since the drill hole is set to `3.2mm` it can be manufactured from `3.12mm` to `3.33mm`, so it should always fit the `3.00mm` spacer.

hmngwy, 2023-01-31:
I see, I did suspect I am getting the low end of that manufacturing tolerance, my PCBs read 3.1mm at that hole using a caliper.

I do have those knurled standoffs, as well as hex ones, however, they have manufacturing a tolerance of +/-1mm, and my standoff's diameter 3.1mm (splitkb.com also lists theirs as 3.1mm and not 3.0mm).

I think for wider compatibility with various sources of these spacers, and to accommodate the drill hole tolerances, it may still be worth adding .1mm to increase the chances of compatibility.

duckyb, 2023-02-01:
Fixed in 31fb1d708d5e074c8c72d372f89e7fc1b5d0feb9

closed 2023-02-01 by duckyb

matru, 2023-02-02:
@duckyb @hmngwy 

I know that on splitkb the diamter is 3.1mm on the knurled cylindrical ones, but every top search I did with standoff/spacer keyword on aliexpress, was 3.2mm diameter for knurled cylindrical ones, 3.3mm max diagonal for hex. Actually could not find something else then these dimensions for an m2.

https://www.aliexpress.com/item/1005001478301407.html
https://www.aliexpress.com/item/1005003106030065.html
https://www.aliexpress.com/item/1005001479466801.html
https://www.aliexpress.com/item/1005001478301407.html
https://www.aliexpress.com/item/1005001479466801.html
https://www.aliexpress.com/item/32841023756.html
https://www.aliexpress.com/item/1005002467410552.html
https://www.aliexpress.com/item/32863484622.html

duckyb, 2023-02-02:
Thank you for your effort @matru 

Now the holes have been changed to 3.35mm which should be enough for the brass spacer. The hex spacer is out of scope for this issue.

matru, 2023-02-03:
@duckyb thanks! What hole diameter does the other variants use, especially the Sweep v2+?

duckyb, 2023-02-03:
> @duckyb thanks! What hole diameter does the other variants use, especially the Sweep v2+?

The Sweep v2 does not have holes for standoffs

matru, 2023-02-04:
@duckyb oh, right. This is not something intentional I assume, there has been simply no development on it?

duckyb, 2023-02-04:
@matru yes. Since it doesn't have a plate case the standoffs are not needed.