davidphilipbarr/Sweep #69: Custom PCB design doesn't work

Sheape, 2023-03-26:
Believe it or not, it's been 11 months since I ordered all the parts and everything but I still haven't fixed my keyboard since then. Before I would solder the key switches, connector, and pro micros, I want to test if every keys work as intended (based from my layout) but none of the keys register. However, I tried to move around a thin wire around the pro micro and it does register the keys which means the problem isn't from my keyboard layout/firmware. Im not really sure if it's my pcb that's causing the problem, when I ordered it from JLC, they checked it and everything is fine. Please if anyone can verify if my pcb is correct, I will appreciate it.

[ferris-sweep-edge-cut-master.zip](https://github.com/davidphilipbarr/Sweep/files/11070724/ferris-sweep-edge-cut-master.zip)

duckyb, 2023-03-26:
Is it a Sweep Half Swept or Sweep Bling MX?

davidphilipbarr, 2023-03-26:
the kicad files look ok, so pix plz!

Riccardo-merli94, 2023-03-29:
I’m currently having the same issue using the half swept. 
I’m not an expert but my understanding is that one pin of the switches should always go to ground, in my pcb all the switches had a ground connection except for one and it looks like that one brings GND to the microcontroller as well. 
my first thought was that was a mistake on the design and I connected the ground with a wire. It did work, when connected via usb, but I’m having issues if I connect the battery. I checked for shorts or wrong connections but I couldn’t find any. The only charge I made was to the outline to change the cut.

duckyb, 2023-03-29:
@Riccardo-merli94 please check the schematic pdf available in the repository and make sure your pcb matches the schematic. The Sweep is a very well tested design so I think we can rule out any issue with the PCB design.

Riccardo-merli94, 2023-03-30:
That’s exactly what I thought, I’ll do some more digging. In case I can’t figure it out, would you be able to give me a little help?

duckyb, 2023-03-30:
> @duckyb That’s exactly what I thought, I’ll do some more digging. In case I can’t figure it out, would you be able to give me a little help? 

I can't promise anything, but we'll see

Sheape, 2023-03-30:
> the kicad files look ok, so pix plz!
> […](#)

damn, thanks for checking it. I'll provide some images of my setup and my layout (just in case that is the problem) later on. Im not really sure what's wrong tbh

Riccardo-merli94, 2023-03-31:
> > @duckyb That’s exactly what I thought, I’ll do some more digging. In case I can’t figure it out, would you be able to give me a little help?
> 
> I can't promise anything, but we'll see

It turned out it was my fault 🤦🏻‍♂️, when I edited the outline for the cutout I accidentally move the edge too close to the power switch and this caused to isolate the right side with the left side. I believe that I can fix it by jumping the two sides with a cable for now, but I’m not 100% sure 😅.

closed 2023-04-09 by duckyb