davidphilipbarr/Sweep #70: How to build a gerber (wanted to add custom silkscreen) for KiCad 7?

vladimirlucifer, 2023-03-26:
Why hello! I wanted to get a PCB of the Sweep v2.2 on jlcpcb and the provided gerber file works just fine (as in, it displays just fine)! However, when I try to add a custom Silkscreen and try to plot it (on KiCad 7) with the website's recommended setting, it does not work (as in, it does not display). What are the settings used to get the provided gerber, so that I can also be sure it is 100% compatible? Many thanks. I also think this may be related to some mising libraries, but I would not know how to properly check and fix that, as I am an absolute noob when it comes to KiCad and other CAD software.

duckyb, 2023-03-26:
The gerber preview viewer from JLC is super unreliable, don't stress about it. A JLC engineer will have to check the file regardless and they will email you if there are any apparent problems.

vladimirlucifer, 2023-03-26:
Thanks! I ended up committing the sin of just substituting the silkscreen layers, and letting everything else be as provided. But with those news, I feel safer and more confident, thank you!

closed 2023-03-26 by duckyb