davidphilipbarr/Sweep #73: any interest in a diodes sweep?

idank, 2023-07-25:
Hello,

I took the sweep pcb and added diodes to it so the SDA/SCL pins can be freed for things like a trackball or OLED screen. I also made it hotswappable while I was at it. Apart from that, it's identical to the OG sweep.

Any interest in adding this variant to this repo?

[image]

duckyb, 2023-07-25:
I don't think it's a good idea to add it to the repo because using diodes is not in the spirit of the sweep. Also since you are using a different pin-out of the micro controller this would be the only sweep that is not compatible in firmware.

I also made something similar to this but it has it's own repo: https://github.com/duckyb/urchin

idank, 2023-07-25:
It doesn't have to use the same firmware, it could sit under `ferris/sweepdiode` or whatever. But the rest makes sense so I'll keep this separate, thanks!

closed 2023-07-25 by idank

duckyb, 2023-07-25:
I know it doesn't _have_ to use the same firmware, but all other sweeps do. So this one would be the odd one out.