davidphilipbarr/Sweep #74: J1 Jumper Issue

jdc75, 2023-08-12:
Hi,

basically I am facing the same issue addressed in #55 but the measurements given did not resolve it for me. I tested for continuity on the VCC net as well as switch 18 net as indicated by @duckyb which were fine. I added both `#define SPLIT_USB_DETECT` as well as `#define SPLIT_USB_TIMEOUT 2500` to my config and flashed both sides. They are working when only one of them is connected to the computer. 

While testing for functionality I noticed an unexpected behavior - at least for me. When one side e.g. right hand is connected to the computer and a TRRS cable is conneted to it which is **not** plugged in to the other hand the right hand stops working. Is this how it should be?

I stumbled across an [article](https://docs.splitkb.com/hc/en-us/articles/360010588860-Only-one-half-of-my-keyboard-works-at-a-time-but-not-when-they-are-both-connected) which is stating that this could be related to J1 Jumper pad found on some pro micro mcs - same goes for mine. Has anyone here experience on this?

duckyb, 2023-08-12:
Which micro controller are you using? Are you sure the cable is TRRS and not TRS? Are you hot-plugging the TRRS cable?

jdc75, 2023-08-12:
Thanks for the quick answer.

I am using the [diymore Pro Micro](https://www.amazon.de/gp/product/B0BKGSVX2X/ref=ox_sc_act_title_1?smid=A2OYQMAUQB5NOO&psc=1). For the cable I am using [this one](https://www.amazon.de/dp/B01I0SI1SG?ref=ppx_yo2ov_dt_b_product_details&th=1). I feel a little bit ashamed to admit, that I have bought the wrong cable as it seems (product page states TRS somewhere) even though I read the information on the bill of material. I ordered a new cable which is TRRS for sure. I will update you once it arrives.

I generally avoid hot-plugging to not damage my mcs.

duckyb, 2023-08-13:
@jdc75 nothing to be ashamed, it's a common mistake. You can tell the type of cable by counting the metal parts at the end. Your cable has TIP RING SLEEVE, while a TRRS cable would have TIP RING RING SLEEVE.

I'll consider this resolved for now, you can reopen it later if necessary.

closed 2023-08-13 by duckyb