davidphilipbarr/Sweep #77: No Input registered when plugged in on Boot

jdc75, 2023-09-27:
Hi,

when I leave the Keyboard plugged into my Laptop on boot it will not register input by the keyboard. When I replug it everything works fine - even while still remaining on the log in screen. I am running Arch Linux - in case.

Thanks for your help
Jan

duckyb, 2023-09-27:
This could be a problem related to the micro-controller or a configuration of QMK/ZMK or your system. It's unlikely to be related to the Sweep hardware (which is what this repository is about).

jdc75, 2023-09-28:
FYI: this issue is related to GRUB (bootloader on linux). I fixed it by adding `#define SPLIT_WATCHDOG_ENABLE` to my config file. For further information see https://github.com/qmk/qmk_firmware/issues/21264.

duckyb, 2023-09-28:
> FYI: this issue is related to GRUB (bootloader on linux). I fixed it by adding `#define SPLIT_WATCHDOG_ENABLE` to my config file. For further information see [qmk/qmk_firmware#21264](https://github.com/qmk/qmk_firmware/issues/21264).

Thank you for sharing 😃