davidphilipbarr/Sweep #79: Question: Switch compatability

SolninjaA, 2023-11-04:
Hi, I just have a quick question regarding the switch compatibility on the Ferris Sweep boards. I'm looking at the compatibility table, but I don't know if I fully understand it.

Am I correct in assuming that the Sweep2.x, Sweep Bling LP, Sweep High and Sweep Half Swept only support Choc v1 switches (except for the Sweep High, which supports multiple)?

Furthermore, am I correct in assuming the Sweep Bling MX, Sweep High and Sweep High M are the only models which support MX style switches (e.g. Gazzew Boba LT, Gateron Milky Yellows)?

Also, I had a look at an interesting switch but I'm not sure if it's compatible with any of the models. The switch I am talking about is: https://www.lofree.co/products/lofree-flow-the-smoothest-mechanical-keyboard?variant=44226590474459

If anyone could help me with these questions, I would greatly appreciate it.
Thanks.

duckyb, 2023-11-05:
All your assumptions are correct, I don't know if that switch is compatible.

SolninjaA, 2023-11-06:
Thanks for replying! I'm surprised that my assumptions are correct :)

I only want and need MX-style switches. Since this is the case, do you think the Ferris Sweep Bling MX would be the best? It looks to me like it's the only model with hotswap compatibility.

Also, regarding the "⚠" symbol on the compatibility table for the Sweep Bling MX, is that just saying it won't work with Alps switches but works perfectly with MX switches and that it requires hotswap sockets to be installed to work?

Thanks.

duckyb, 2023-11-06:
Yes I would recommend the bling mx.
Yes that's what the warning symbol is for.

SolninjaA, 2023-11-13:
Great, thanks so much!

closed 2023-11-13 by SolninjaA