davidphilipbarr/Sweep #85: Question; What type/size of Mill-Max Hotswap Sockets?

Vartroc, 2024-03-21:
Hello,
I was just browsing and searching some materials, when the question arose what type/size of Mill-Max Hotswap Sockets would fit. Maybe I just didn't read carefully enough (I tried) but I can't seem to find any information about that.

I would be delighted if anyone could help me.

duckyb, 2024-03-22:
If you are planning for hotswap why don't you build a sweep bling instead?

Vartroc, 2024-03-23:
You are totally right. I was focusing so hard on trying to find the right Mill-Max Socket, that lost my view of the bigger picture...

Vartroc, 2024-03-23:
Just being curious tho, why is there even the option for a Hotswap Sweep if the Bling exists?

closed 2024-03-23 by Vartroc

duckyb, 2024-03-23:
> Just being curious tho, why is there even the option for a Hotswap Sweep if the Bling exists?

You mean why is there an option to add millmax sockets to any other sweep? It's because:

1. That option came before the sweep bling was designed
2. To help people that already have a sweep and would like to try hotswap on a budget