davidphilipbarr/Sweep #86: Preview not loeading on JLCPCB

adnanaq, 2024-08-19:
Hi,
I am trying to place fabrication order on JLCPCB for sweep bling lp. However, when I load the zip in jlcpc, it doesn't show any image preview of the pcb and doesn't load the dimension. And when trying to add into the after adding the dimensions and required settings manually, it returns `Paramter Exception` error. 

[image]

adnanaq, 2024-08-19:
Tried with Sweep High. Same issue.

adnanaq, 2024-08-19:
Figured it out. I had to run Plot so it could autofill zone and also the drill files are missing from the gerber zip.

jimmyff, 2025-10-01:
Hey @adnanaq I have the same issue, I'm uploading to JLCPCB but all the drill holes appear to be filled in. How did you resolve it?

Thanks!

adnanaq, 2025-10-01:
Also https://forum.kicad.info/t/drill-holes-dont-show-when-printed/21003