davidphilipbarr/Sweep #91: Why only TRRS cable?

jakobkhansen, 2025-02-27:
Hi, I'm looking to understand why I cannot use a TRS cable with this board. From my understanding and from a QMK perspective, if you use a serial communication driver, it will only use three pins on the TRRS jack, and therefore it should work with a TRS cable. I have a Corne keyboard which works fine with a TRS cable, so I'm wondering what the difference is for the Sweep? I have not had any luck with using a serial driver and a TRS cable, only TRRS cable works.

gromms, 2025-05-07:
I was wondering the same thing but it looks like the pcb is designed so that TRS cable would short Vcc straight to ground so make sure you never use TRS cable when you have the board powered up. There does not seem to be any reason why it should be designed this way and you are right, only three pins of TRRS are used. 

If the pins were swapped so that the S and R2 were GND or if S was not connected anywhere, then TRS cable should also work.

davidphilipbarr, 2025-05-08:
It's a fair point, I think the idea was to theoretically reduce the chance
of shorting. If someone submitted an update I'd apply it but other than
that it is what it is now I guess :)

duckyb, 2025-05-08:
Turned this question into a feature request, anyone is welcome to work on this.